Flipper-XFW/Xtreme-Firmware #217: Bluetooth not connecting discretely anymore

St31th, 2023-04-26:
### Describe the bug.

Before the change to badbt, if my phone was connected to a bt speaker, I could name the flipper the same name and turn off the speaker and my phone would automatically connect to the flipper. Now it doesn’t and no matter what it always pops up a request to connect message on my phone now. 

### Reproduction

Just connect to badkb using Bluetooth on a phone

Willy-JL, 2023-04-26:
did you try toggling the remember option, on both modes?

also, if it worked that way before, that is not an intended feature of badkb. that *should not* happen. if it did, that is some serious security vulnerability on your phone my friend

ClaraCrazy, 2023-05-05:
@

closed 2023-05-05 by ClaraCrazy