Flipper-XFW/Xtreme-Firmware #224: Mute/unmute flipper with one click

rhl2401, 2023-05-01:
### Describe the enhancement you're suggesting.

I like to have the Flipper make sounds while interacting with it, but sometimes the sounds are really not appropriate (especiallialy places where they are very audible). 

It would be nice to be able to click on the volume bar in the notification area to mute/unmute the device altogether. Today, you must click "down" 20 times while having it make 20 beeps on different volume levels until it's completely silent. 

### Anything else?

You could always make the device remember the latest volume setting and restore to that one, but that is not really a concern for me. The ability to go from 100% to 0% and back to 100% would do the trick for me.

Willy-JL, 2023-05-01:
Already exists in dev branch

rhl2401, 2023-05-01:
Fantastic. I didn't realize steath mode included this feature. Thank you for your work!

closed 2023-05-03 by github-actions[bot]