Flipper-XFW/Xtreme-Firmware #228: "stop emulating favorites after X seconds" causes crashes on Security+ 1.0 SubGhz

OptimalN0body, 2023-05-03:
### Describe the bug.

After setting the "stop emulating favorites after X seconds" setting in the Xtreme settings, the ability to send the SubGhz signal with the center button was gone, and was sending the signal continuously. Then the flipper crashed with a 'NULL pointer dereference' error.

### Reproduction

1. Have a Security+ 1.0 Subghz rolling code signal set up
2. Change the Favorites Timeout to 10s
3. Run the  Security+ 1.0 Subghz rolling code signal
4. Experience crash

ClaraCrazy, 2023-05-03:
Hi there.

I just tested the feature with normal cards & chips and it works fine. It *might* be subghz or its rolling code protocols. Unfortunately thats nothing I can test. 

Are you able to provide either a copy of your signal, or full debug logs via a wifi devboard?

Willy-JL, 2023-05-03:
Was able to replicate. As for the button missing and it transmitting for a long time, you're missing the point of the setting... it keeps transmitting that signal for the time you decide, then after that time it stops and goes back to the favorites. There is a crash however, which is if you quit before it does so by itself.

OptimalN0body, 2023-05-03:
Yes, I understand that it removes the button and just Auto-plays the signal.
It might be nice to be able to pick which types of files are auto-played and then timeout, because I would definitely want the Timeout on my RFID work badge, but not on my Garage door opener. I know this is a brand new feature, but this is just a suggestion for possible future improvements.

closed 2023-05-04 by github-actions[bot]