Flipper-XFW/Xtreme-Firmware #232: Bluetooth Remote Bugged

xAstroBoy, 2023-05-04:
### Describe the bug.

The Youtube Shorts / Tiktok Bluetooth remote doesn't work properly

On Iphone all it does is close the app or control the audio, doesn't pause the short or scroll down on button click.

### Reproduction

Just use the remote in BT Remote and you will see it doesn't behave properly.

fred-corp, 2023-05-04:
Same issue here; When I open the Bluetooth Remote app, the Flipper crashes with a NULL pointer dereference message after reboot.

Willy-JL, 2023-05-04:
@fred-corp thats not the problem this issue talks about... that abounds to me like your app, or your firmware, didn’t update correctly. Please reinstall following the instructions

ClaraCrazy, 2023-05-04:
iPhone moment. Cant change it, if those, for no reason as ususal, expect different data than everything else in existence. Remote works mostly (Shorts is still in beta and bc written by a weirdo, device specific) fine where designed to, so just buy proper devices that respect normal bt instructions.

xAstroBoy, 2023-05-04:
> 

TikTok almost worked before in previous builds, somehow this build messed it up :C

ClaraCrazy, 2023-05-11:
Well, its not our issue either way, its the issue of whoever wrote the app.

closed 2023-05-11 by ClaraCrazy

fred-corp, 2023-05-11:
Seems fixed in 0046 anyway..