Flipper-XFW/Xtreme-Firmware #237: cannot update to 0046 - Error need more free space in - internal storage

TheMany172, 2023-05-06:
### Describe the bug.

Error
Need more free space in - 
internal storage

### Reproduction

downloaded 0046 via https://github.com/ClaraCrazy/Flipper-Xtreme/releases/tag/XFW-0046_06052023

was updating from 0045 (after hotfix)

copied accross from computer using qflipper

browse to update on flipper and press run in app

[image]

Willy-JL, 2023-05-06:
what files do you have on internal storage?

TheMany172, 2023-05-06:
Not too sure tbh - I have never added anything to it. When I check via qflipper it comes back with just some hidden files
[image]

Willy-JL, 2023-05-06:
yeah... RM moment... delete dt settings, passport settings, and if still not working delete region data too.

only needed files are the u2f ones and notification settings, region data if you care about region lock, remove if you want

TheMany172, 2023-05-06:
Ah ok - so an issue due to me previously having RM as firmware then? thats a shame

Willy-JL, 2023-05-06:
well, those files are there (mainly passport settings) because of RM, then our firmware is getting too big, so it cant fit those files + our firmware. xtreme doesnt use those, so just delete them

TheMany172, 2023-05-06:
Np - I deleted the first two you said (bt settings, passport settings) but it still wouldnt, so also removed region data
ran update - it did a flipper has crashed null point reset blah
ran again - all good

Thanks Willy, glad its just an unrelated issue

closed 2023-05-06 by TheMany172