Flipper-XFW/Xtreme-Firmware #241: Reader attack reports incorrect keys

DidierA, 2023-05-11:
### Describe the bug.

Using latest Xtreme firmware, NFC Reader attack reports incorrect keys.
This is due to a wrong cuid being written in nfc/.mfkey32.log
I could pinpoint the issue by emulating a reader with a proxmark3, using the flipper in Reader attack mode and looking at the trace. The UID sent by the flipper is not the one it dumps in the log file.

### Reproduction

+ On flipper, go to NFC/Reader Attack
+ use a proxmark3 to emulate  a reader with the  lua script provided below, or any other means to read a block with a known key, and report the UID given by the flipper.
+ Press the flipper to the proxmark
+ with qflipper app, download `nfc/.mfkey32.log`
+ with the flipper phone app, launch mfkey32v2, the found key should be A0A1A2A3A4A5 and FFFFFFFFFFFF (if using the lua script without parameters), note that they are not.
+ In `.mfkey32.log`, note that the cuid is not correct: it should be the last 4 bytes of the UID reported by the proxmark.
+ Optionaly, change the cuid field to the correct value in each line, upload to flipper again, launch mfkey32v3 attack, and note that the keys are correct.

### Anything else?

lua script to use on the proxmark:
```lua
local getopt = require('getopt')
local lib14a = require('read14a')
local cmds = require('commands')
local utils = require('utils')
local ansicolors  = require('ansicolors')

copyright = ''
author = "Didier"
version = 'v1.0.4'
desc = [[
This is a script which simulates a mifare classic reader. It sets itself into
'listening'-mode, after which it continuously tries to read a specific sector
on any mifare classic card that you place by the device.

based on hf_mf_autopwn.lua
]]
example = [[
    1. script run hf_mf_reader
]]
usage = [[
script run hf_mf_reader [-h] [-d] [-a <key>] [-b <key>]
]]
arguments = [[
    -h         this help
    -d         debug logging on
    -a         key to use, keytype A
    -b         key to use, keytype B

]]

-------------------------------
-- Some utilities
-------------------------------
local DEBUG = false
---
-- A debug printout-function
local function dbg(args)
    if not DEBUG then return end
    if type(args) == 'table' then
        local i = 1
        while result[i] do
            dbg(result[i])
            i = i+1
        end
    else
        print('###', args)
    end
end
---
-- This is only meant to be used when errors occur
local function oops(err)
    print('ERROR:', err)
    core.clearCommandBuffer()
    return nil, err
end
---
-- Usage help
local function help()
    print(copyright)
    print(author)
    print(version)
    print(desc)
    print(ansicolors.cyan..'Usage'..ansicolors.reset)
    print(usage)
    print(ansicolors.cyan..'Arguments'..ansicolors.reset)
    print(arguments)
    print(ansicolors.cyan..'Example usage'..ansicolors.reset)
    print(example)
end
---
-- Waits for a mifare card to be placed within the vicinity of the reader.
-- @return if successful: an table containing card info
-- @return if unsuccessful : nil, error
local function wait_for_mifare()
    while not core.kbd_enter_pressed() do
        res, err = lib14a.read()
        if res then return res end
        -- err means that there was no response from card
    end
    return nil, 'Aborted by user'
end

---
-- The main entry point
local function main(args)

    local res, uid, err, _, sak
    local seen_uids = {}
    local keyA = 'A0A1A2A3A4A5'
    local keyB = 'FFFFFFFFFFFF'
    local cmd_template = 'hf mf rdbl --blk 0 %s -k %s'

    -- Read the parameters
    for o, a in getopt.getopt(args, 'hda:b:') do
        if o == 'h' then help() return end
        if o == 'd' then DEBUG = true end
        if o == 'a' then keyA = a end
        if o == 'b' then keyB = a end
    end

    print('Waiting for card or press Enter to stop')
    res, err = wait_for_mifare()
    if err then return oops(err) end
    
    uid = res.uid
    sak = res.sak

    print(string.format('Card found. UID: %s', res.uid))

    print (string.format('Reading continuously with keyA: %s and keyB: %s. Press Enter to stop', keyA, keyB))
    local steps=6
    while steps > 0 do
        steps = steps -1 
        local cmd = string.format(cmd_template,'-a', keyA)
        core.console(cmd)
        core.console('trace list -t mf')
        cmd = string.format(cmd_template, '-b',  keyB)
        core.console(cmd)
        core.console('trace list -t mf')
    end

end

-- Call the main
main(args)
```

ClaraCrazy, 2023-05-11:
Yea, we know. Someone else already mentioned it like two days ago. Not sure where it comes from tho unfortunately (yet)

ClaraCrazy, 2023-05-13:
fixed in next release

ClaraCrazy, 2023-05-29:
fixed in release

closed 2023-05-29 by ClaraCrazy

DidierA, 2023-05-30:
Thank you, tested this morning, I can confirm it is now working as expected.