Flipper-XFW/Xtreme-Firmware #246: Crash on idle - null pointer dereference

TheMany172, 2023-05-13:
### Describe the bug.

In response to notice on discord - firmware-updates asking to be told if this happens

happy to help, if that is possible

### Reproduction

not sure exactly how to recreate or what info would be of use

I turned it on an hour or so ago - when i came  back to my desk it had the

'flipper crashed but has been rebooted, null pointer dereference'

ClaraCrazy, 2023-05-13:
Please message us on discord. The issue is something we technically arent even allowed to publicly talk about (yet), not caused by firmware. Thats all I am able to say here unfortunately.

closed 2023-05-13 by ClaraCrazy

TheMany172, 2023-05-13:
oh ok, fair enough - will close

CapuchaRojo, 2023-05-13:
Yes ma’am there is more. I may flash things wrong.
iOs..

Sometimes when running the apps straight from your new gooey system wii icons, Wi-Fi folder, I will only generate the 160+ pages  log of repetitive bad heading error.

I have Apple devices
Own 13promax and iPad gen 10.
I have access to a MacBook which makes life easier with aurduino and terminal and the usually reliable GitHub web updaters.

But I’ll let you see what iOS is doing to me: pics attached
I’ve spoke with skelli briefly on it. All his stuff worked except step 0 for mac, but I got it to work anyhow somehow with his release. 🤷‍♂️ I’m just going thru all the available avenues to flash my boards.

I have multiple issues but figured I would start with this one.

Thank you for helping me btw
It is much appreciated. And since not in discord with ya anymore I know u doing this is beyond. So after done, I will shoot u patreon style loving for sure.

Thanks again

Sent with Spark

ClaraCrazy, 2023-05-13:
*confused clara noises*

CapuchaRojo, 2023-05-13:
Basically
I couldn’t get the flash for the life of me and when I finally did it didn’t do all the steps correctly 100%. There is a step zero for Mac that I couldn’t get accomplished. It was a download for -r required

But I was told I was over complicating it by Skell and that if it’s not found it’s not found

But then I can not run the maurader without getting the errors in the pics.
Error- invalid header: 0xe0ffd8ff
Ran 1165 times.

Sent with Spark

ClaraCrazy, 2023-05-13:
has nothing to do with the issue.

CapuchaRojo, 2023-05-13:
Well. Ok. Giving u possible issues that come before it

Sent with Spark

CapuchaRojo, 2023-05-17:
Messaged to CC but couldn’t find themany172

JustThisGuyJoe, 2023-05-18:
Hey Clara, I capture some profiles for an overhead fan for you to include on your next release. 
https://github.com/JustThisGuyJoe/RF-Profiles Please don't cite me or credit me in any way, just throw them in there, I'm a private person who values my anonymity.