Flipper-XFW/Xtreme-Firmware #248: Favourite app (Directories)

HyperMuffin12, 2023-05-14:
### Describe the enhancement you're suggesting.

I use my Flipper as a password manager, so I wondered if I could quickly access my password directory with Fav App (Up) / (Down) feature. Turns out I can't. So I was figured it would be worth suggesting as it would save time on opening BadKB with Fav App and then finding my password directory.

ClaraCrazy, 2023-05-14:
*huh*

ClaraCrazy, 2023-05-14:
you can just mark a directory as favorites in the file browser? Not to mention that hearing "BadKB" in combination with "Password manager" sounds about as wrong as Roguemaster and original code.

HyperMuffin12, 2023-05-14:
> *huh*

I mean automatically opening a directory within an app I guess. Instead of opening just BadKB with Fav App, it would open BadKB and then a chosen directory.

Willy-JL, 2023-05-14:
Wrong idea. If we do this, we make it so you can set a folder as favorite and it opens it in the *archive*. Opening in the appropriate app is not at all feasible.

But either way, your suggestion makes sense, your reasoning for it *does not*, in any way shape or form. Veeeery bad idea to do passwords with badkb.

alecbramham, 2023-05-16:
> > Wrong idea. If we do this, we make it so you can set a folder as favorite and it opens it in the _archive_. Opening in the appropriate app is not at all feasible.
> > But either way, your suggestion makes sense, your reasoning for it _does not_, in any way shape or form. Veeeery bad idea to do passwords with badkb. Please do yourself a favor and dont do that
> 
> I appreciate your concern. Could you tell me why using BadKB for my passwords is a bad idea

You're sending sensitive data over a completely unencrypted Bluetooth signal. And if the BadKB payload navigates to the site or names the passwords, anyone who can intercept it knows what the password unlocks.

HyperMuffin12, 2023-05-16:
> You're sending sensitive data over a completely unencrypted Bluetooth signal. And if the BadKB payload navigates to the site or names the passwords, anyone who can intercept it knows what the [...]

Valid points. However, I only use USB to enter the passwords and the payload itself is literally just "STRING mypassword"

Willy-JL, 2023-07-04:
with the new keybinds system this might be more doable, but still misses a way to select the folder, as the file picker only picks files. maybe next release will include this

Willy-JL, 2023-08-10:
now partially implemented. can select files for now. for folders it still needs a new file browser option to enable selecting folders