Flipper-XFW/Xtreme-Firmware #249: bluetooth not staying conected

seeingoblivion, 2023-05-14:
### Describe the bug.

When using Bluetooth remote, when doing a fresh connection, it works perfectly, but when exiting completely, not just choosing a different function. it and the computer will say still connected, but will not work again unless re-paired. 

### Reproduction

Pair flipper to computer (control not regular)
Enter Bluetooth remote
Works fine
Exit to menu and choose different function, works fine
Exit Bluetooth remote completely , and reenter
Will reconnect, but will not work on any function
Then unpair and forget connection
Then reconnect and works fine again

### Target

latest release

Willy-JL, 2023-05-14:
fixed in dev branch

seeingoblivion, 2023-05-14:
Ah perfect, thanks for the quick response!

ClaraCrazy, 2023-05-29:
fixed in release

closed 2023-05-29 by ClaraCrazy