Flipper-XFW/Xtreme-Firmware #250: Learning new IR remote always open “raw” mode.

Przemekmac1, 2023-05-16:
### Describe the bug.

When I go infrared/learn new remote it will open app for learning in raw mode, also when I selected normal mode. 

### Reproduction

1. Switch on
2. Go IR/learn new remote (non-raw).

ClaraCrazy, 2023-05-16:
lmao. how tf did we fuck that up? consider it fixed next release

Willy-JL, 2023-05-21:
After looking into it, this is intended behavior. Learn remote raw disables trying to decode the signals, while learn remote keeps decoding enabled. Thing is, if the IR samples it picks up in normal learn remote do not correspond to any known protocol it knows how to decode, it still picks them up as raw. Learns raw simply stops it from *trying* to decode

closed 2023-05-21 by Willy-JL