Flipper-XFW/Xtreme-Firmware #251: EPIRB emergency beacon decoder.

Przemekmac1, 2023-05-16:
### Describe the enhancement you're suggesting.

I think it would be usefull to add 406.025MHZ BPSK decoder for signal transmitted by EPIRB (ELT) beacons.  It could be used for testing and analyze of data they are sending.  
I could provide some knowledge and also testing on this subject however I’m not perfect in programming ;). 

I have some materials and also example C program code. 

https://www.cospas-sarsat.int/en/beacon-decode-program

https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.633-1-199006-S!!PDF-E.pdf

https://www.techrxiv.org/articles/preprint/Implementation_of_a_GNU_Radio_Based_Search_and_Rescue_Receiver_on_ESA_s_OPS-SAT_Space_Lab/14992086/files/28858377.pdf

https://www.coaa.co.uk/epirbplotter.htm

sebdemars13, 2023-10-25:
Bonjour,

Avez vous réussi à intégrer le protocole epirb dans portapack mayhem ?
Merci pour votre réponse
Cordialement

Przemekmac1, 2023-10-25:
> Bonjour,
> 
> 
> 
> Avez vous réussi à intégrer le protocole epirb dans portapack mayhem ?
> 
> Merci pour votre réponse
> 
> Cordialement

Jak narazie nikt się tym nie zajął.

sebdemars13, 2023-10-25:
Merci pour votre réponse. J’espère que ce projet sera fait.
Cordialement