Flipper-XFW/Xtreme-Firmware #252: Update #046 failed

8um1c4, 2023-05-17:
### Describe the bug.

When trying to update from 45 to 46 via web, I get the following message:
Update Failed!
Error #[1-20]

Any help? 

### Reproduction

I've tried 4 times.

Willy-JL, 2023-05-17:
what files do you have on internal storage? (enable hidden files in qflipper)

8um1c4, 2023-05-17:
Only 5 small files. Probably there are files hidden. Can you please help how do I un-hidden them?

Willy-JL, 2023-05-17:
Its not about the files being small. I asked what files you see, 5 files sounds normal, mind telling me what they are?

Willy-JL, 2023-05-17:
delete all of em then retry. maybe keep dolphin state if you want, others are fine to lose (just small setting options)

8um1c4, 2023-05-17:
Solved ! Many thanks Willy-JL! Never happened before!
Regards

closed 2023-05-17 by 8um1c4