Flipper-XFW/Xtreme-Firmware #256: Crashes when exiting CLI (subghz chat)

Kuolinisku, 2023-05-19:
### Describe the bug.

When you exit the CLI application initially it is fine, and you can also back out to the apps menu, but when you hit back for a 3rd time to return to the home screen it crashes with a "NULL pointer dereference" error.  

### Reproduction

1. Go to Apps > Tools > CLI (subghz chat)
2. Launch the app
3. Long-press the BACK button to exit
4. Press the BACK button to go back to the Apps menu
5. Press the BACK button to go back to the home screen
6. This is where the error should occur if it behaves the same as mine is

### Anything else?

Fairly minor issue - it recovers immediately from the crash and you were exiting everything anyway - but annoying for people who use the app probably.

Willy-JL, 2023-05-19:
Fixed in dev branch

Kuolinisku, 2023-05-19:
Thanks!  

Is there a spot I should have checked for that?  I did skim the existing bugs but that is all I checked.  Anyway good to know it is fixed.

Willy-JL, 2023-05-20:
well in this case it wasn't reported by a user so its not in an issue, nano (from unleashed) informed us and gave a fix for it. it is listed in the projects tab in the done column tho

ClaraCrazy, 2023-05-29:
fixed in release

closed 2023-05-29 by ClaraCrazy