Flipper-XFW/Xtreme-Firmware #257: App and bluetooth not working

luqezr, 2023-05-20:
### Describe the bug.

Hi All, since i’ve installed this firmware i’ve lost the ability connect via bluetooth to the flipper with the app to controle it remotely.
If I activate bluetooth my device wont show up, I've tried with iOS, android, windows and linux but nothing :(.

If I use the badusb via bluetooth it works, but thats the only bluetooth feature i can use, could it be that this affects the normal bluetooth use?

I've reinstalled the firmware a few times

### Reproduction

1 open bluetooth
Flipper zero wont show up, but honestly there is no error sign

Willy-JL, 2023-05-20:
you need to use the flipper mobile app to pair

luqezr, 2023-05-20:
> you need to use the flipper mobile app to pair

Hi Willy, thanks for the quick reply, it worked! 
It synchronizes, but if I try to emulate something from the app my flipper doesnt do anything

Willy-JL, 2023-05-20:
already working on a fix for that, mind pinging me in the discord so i can add you to the ticket?

luqezr, 2023-05-20:
> already working on a fix for that, mind pinging me in the discord so i can add you to the ticket?

Hi willy, sure, i'll do that :)

Willy-JL, 2023-05-20:
confirmed fixed in dev branch via discord, will be in next release

sariyan-0, 2023-05-21:
Same i have the same problem 😕

Willy-JL, 2023-05-21:
> Same i have the same problem 😕

Well read the messages above. 1: it won’t show in Bluetooth, it shows in the mobile app. 2: emulating might not work, and this is fixed in dev branch and will be in next release

ClaraCrazy, 2023-05-29:
fixed in release

closed 2023-05-29 by ClaraCrazy