Flipper-XFW/Xtreme-Firmware #258: Flipper zero passport/status

sariyan-0, 2023-05-21:
### Describe the enhancement you're suggesting.

Hi there i have a little issue with the xtreme flipper zero OS and that's because my dlipper is always angry on the passport and idk why because i use it so many times like i use it it read and emulate stuff but the status never changes i tried factory resetting it and everything reinstalling the OS but still it dosent do anything i wanna know how can i fix it and is there anything i should change in the settings to fix it?

Willy-JL, 2023-05-21:
What happened is that butthurtness (flipper becoming angry) has never worked well on any firmware, but we fixed it, so now it does become angry if you don’t use it enough. You can change how often it becomes more angry in xtreme settings > misc, there’s a butthurt timer. If you want it to always be happy, turn it off and then over time it will become happy again

closed 2023-05-21 by Willy-JL

sariyan-0, 2023-05-21:
So does it mean for example if i set the timer to 30 minutes like does it
mean that if i dont use it for 30 minutes it becomes angry right or
something else?

And btw how long does it take to be happy again because i want it to be
like a tamagotchi so it becomes mad if i dont use it much.

So can you please tell me what is the best time to put it and what exactly
the time means with an example u would really appreciate it thanks.

Willy-JL, 2023-05-21:
Next release makes this more clear with an additional entry in the xtreme app, for "dolphin angry" which shows how this works internally. it's a number value from 0 to 14, 0 is most happy 14 is most angry. butthurt timer is simply how often it will increase the number, while using the flipper descreases the number.

sariyan-0, 2023-05-21:
For example if u set it to 6 hours dose it mean that itz going to be mad for every 6 hours i dont use it or its going to be mad for 6 hours?

Willy-JL, 2023-05-21:
read what i said another time, just slower

sariyan-0, 2023-05-21:
Okay got it! But does it become angry due to inactivity or smth?

Btw when is the nest release date?

Willy-JL, 2023-05-21:
> Okay got it! But does it become angry due to inactivity or smth?

once again, you can answer this with what i already said, guess ill repeat it. the butthurt timer increases the butthurt value, always does it, at the interval you choose. when you use your flipper it decreases the butthurt value.

> Btw when is the nest release date?

when its ready

sariyan-0, 2023-05-22:
Got it thanks but it has been a few hours and its still angry what should i
do

Willy-JL, 2023-05-22:
keep using it

hack5aw, 2023-05-27:
why did i find this so funny