Flipper-XFW/Xtreme-Firmware #262: Dummy Mode

DarkEnder4583, 2023-05-24:
### Description of the feature you're suggesting.

Contrary to popular belief, I know many people who enjoyed Dummy Mode, including myself. It should be added back, or at least add a setting to enable/disable it.

hack5aw, 2023-05-24:
As another user who also like dummy mode I agree with him.

JTCozart, 2023-05-26:
It is nice if there is someone that is skeptical of your flipper. Dummy is a nice way to say... look it's just a game.

ClaraCrazy, 2023-05-26:
yeah no. People dont ask you what this is. They know.

closed 2023-05-26 by ClaraCrazy

DarkEnder4583, 2023-05-26:
Respectfully, I've actually had many people ask me what it is and have gotten away with it using dummy mode. 
Ngl, considering some of the other apps (quite honestly, some inappropriate) included in Xtreme, I fail to see what's so wrong with dummy mode. I always saw third party Flipper builds as something to add to the firmware, not take away. Would you mind explaining your thought process for removing it?

ClaraCrazy, 2023-05-26:
the thought process is that it adds no useful functionality, and over 200k people agreed. 

Its cool if your friends ask you. Feds wont.

DarkEnder4583, 2023-05-26:
That was not the main point I intended to discuss, but since you're insisting, I was referring to applications like "Orgasmotron." However, it seems you may not fully recognize the inappropriate nature of such an app, and believe it deserves more storage space than a wholesome trick. Given our differing perspectives on how to utilize such a device, I don't see much value in continuing this discussion. Nonetheless, I would like to express gratitude for your efforts in developing this software. Despite any potential personal biases, I do appreciate the hard work you and your team have dedicated to its creation.

ClaraCrazy, 2023-05-26:
Thats once again a community app, not mine. 

What the community makes has nothing to do with me, lol

JTCozart, 2023-05-26:
It's coming really handy for me when I'm trying to sneak it past security at like a concert venue. The layperson in the line asks what's that... In dummy mode I show them...  It's a game. 

Don't be so quick to dismiss use cases. Covert opsec is a thing. Red team does this stuff all the time.

ClaraCrazy, 2023-05-26:
No, they dont ask. They know. its part of their training at this point. god

JTCozart, 2023-05-26:
I can tell you from real world experience than not everybody knows yet!
It's still a useful too for red teams.