Flipper-XFW/Xtreme-Firmware #264: Error while installing firmware

Padra50, 2023-05-28:
### Describe the bug.

When i try to update or install xtreme it has an error but it does install the firmware although when it try open the xtreme tab it said invalid 

### Reproduction

installing or updating xtreme

Willy-JL, 2023-05-28:
im guessing you mean, it says "preload failed, invalid file" or something like that. seems like indeed it failed to complete the update. what files do you have on internal storage? and did you delete the apps folder before installing?

Padra50, 2023-05-29:
Yes i deleted the apps folder before i installed the firmware and i have nothing on the internal flash apart from a hidden file 
[image]

alecbramham, 2023-05-30:
I just had the XFW-0047 update fail. The error message is shown below. This might not be the same error as Padra50 as my Internal Flash Storage still has some settings, region and u2f files.
I made sure to delete my "Apps" folder.
[image]

Willy-JL, 2023-05-30:
delete all the files from internal storage which are not u2f related

alecbramham, 2023-05-30:
At first it looked like the update was working. Bar went all the way across the screen. But then the Flipper shut down and could only be booted in DFU mode. Not exactly sure what the issue is but I think I'm sticking with 0046 for the time being.

Willy-JL, 2023-05-30:
just clic krepair when in dfu, then install 47

alecbramham, 2023-05-31:
That ended up working. Thanks for your help. I'll make sure to remember these troubleshooting steps for future updates.

closed 2023-05-31 by Willy-JL

Padra50, 2023-06-01:
I uninstalled everything non u2f related in my internal storage and and then tried updating xtreme and I didn't get an error message but when I tried open the xtreme settings it came up with the same error so I was going to try and install the stock flipper firmware and then update it again and see what happens but when I tried updating to the stock firmware it game me an error message but still updated after I cancelled. Here is my log:

389 [APP] qFlipper version 1.3.1 commit 8e43ad60 2023-05-23T13:55:55
389 [APP] OS info: Windows 11 Version 22H2 11 10.0.22621 Qt 6.4.2
403 [DBG] Trying serial port FLIP_SIMERETA at \\.\COM3
403 [DBG] Using  serial port FLIP_SIMERETA at \\.\COM3
406 [RPC] Starting RPC session...
485 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
485 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1650 [RPC] Failed to start RPC session: Failed to open serial port: Access is denied.
1650 [REG] Device initialization failed: Protobuf session error: Failed to open serial port: Access is denied.
7387 [BKD] Finalized current operation
13680 [REG] Detected new device: VID_0x483:PID_0x5740
13690 [DBG] Trying serial port FLIP_SIMERETA at \\.\COM3
13690 [DBG] Using  serial port FLIP_SIMERETA at \\.\COM3
13690 [RPC] Starting RPC session...
13741 [RPC] RPC session started successfully.
13741 [RPC] (1) System Protobuf Version START
13807 [DBG] Detected protobuf version: 0.15
13807 [RPC] (1) System Protobuf Version SUCCESS
13807 [RPC] (2) Property Get START
13926 [RPC] (2) Property Get SUCCESS
13926 [RPC] (3) Storage Info @/ext START
13989 [RPC] (3) Storage Info @/ext SUCCESS
13989 [RPC] (4) Storage Stat @/ext/Manifest START
14041 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
14041 [RPC] (5) System Get DateTime START
14093 [DBG] Flipper time skew is -3356 milliseconds
14093 [RPC] (5) System Get DateTime SUCCESS
14093 [RPC] (6) System Set DateTime START
14145 [RPC] (6) System Set DateTime SUCCESS
14145 [RPC] Stopping RPC session...
14147 [RPC] Attempting to unload protobuf plugin...
14147 [RPC] Cannot unload protobuf plugin. This is probably OK.
14147 [RPC] RPC session stopped successfully.
14147 [REG] Registering the device
14147 [DEV] Version: XFW-0046 commit: 075c52ca radio: 1.13.3
14147 [BKD] Current device changed to "Simereta"
14255 [RPC] Starting RPC session...
14295 [RPC] RPC session started successfully.
14305 [RPC] (1) Gui Start ScreenStream START
14359 [RPC] (1) Gui Start ScreenStream SUCCESS
23928 [RPC] (2) Gui Start VirtualDisplay START
23993 [RPC] (2) Gui Start VirtualDisplay SUCCESS
24061 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
24078 [DEV] Full Update @Simereta START
24082 [UTL] Region Provisioning @Simereta START
24082 [DBG] Development hardware region detected, skipping region provisioning...
24082 [UTL] Region Provisioning @Simereta SUCCESS
24082 [UTL] Refresh Storage Info @Simereta START
24082 [RPC] (3) Storage Info @/ext START
24145 [RPC] (3) Storage Info @/ext SUCCESS
24145 [RPC] (4) Storage Stat @/ext/Manifest START
24212 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
24226 [UTL] Refresh Storage Info @Simereta SUCCESS
24479 [ZIP] Uncompressing file with size of 1096538 bytes...
24493 [ZIP] Uncompression finished : No Error
24497 [UTL] Create Path @/ext/update/f7-update-0.83.1 START
24497 [RPC] (5) Storage Stat @/ext/update START
24559 [RPC] (5) Storage Stat @/ext/update SUCCESS
24560 [RPC] (6) Storage Stat @/ext/update/f7-update-0.83.1 START
24613 [UTL] Create Path @/ext/update/f7-update-0.83.1 SUCCESS
24613 [RPC] (6) Storage Stat @/ext/update/f7-update-0.83.1 SUCCESS
24613 [UTL] Verify checksum of 6 items @Simereta START
24614 [RPC] (7) Storage Md5Sum @/ext/update/f7-update-0.83.1/firmware.dfu START
26081 [DBG] File is identical: "/ext/update/f7-update-0.83.1/firmware.dfu"
26081 [RPC] (7) Storage Md5Sum @/ext/update/f7-update-0.83.1/firmware.dfu SUCCESS
26081 [RPC] (8) Storage Md5Sum @/ext/update/f7-update-0.83.1/radio.bin START
26377 [DBG] File is identical: "/ext/update/f7-update-0.83.1/radio.bin"
26377 [RPC] (8) Storage Md5Sum @/ext/update/f7-update-0.83.1/radio.bin SUCCESS
[... 21 lines cut ...]
29797 [RPC] Attempting to unload protobuf plugin...
29797 [RPC] Cannot unload protobuf plugin. This is probably OK.
29797 [RPC] RPC session stopped successfully.
29801 [REG] Device went offline: VID_0x483:PID_0x5740
30519 [REG] Detected new device: VID_0x483:PID_0x5740
30530 [DBG] Trying serial port FLIP_SIMERETA at \\.\COM3
30530 [DBG] Using  serial port FLIP_SIMERETA at \\.\COM3
30530 [RPC] Starting RPC session...
30583 [RPC] RPC session started successfully.
30583 [RPC] (1) System Protobuf Version START
31363 [RPC] Serial connection was lost.
31367 [REG] Device went offline: VID_0x483:PID_0x5740
31367 [RPC] Stopping RPC session...
31367 [REG] Device initialization failed: Failed to get protobuf version: RPC session was stopped with operations still running
31368 [RPC] (1) System Protobuf Version ERROR: RPC session was stopped with operations still running
31368 [RPC] Attempting to unload protobuf plugin...
31368 [RPC] Cannot unload protobuf plugin. This is probably OK.
31368 [RPC] RPC session stopped successfully.
151688 [REG] Detected new device: VID_0x483:PID_0x5740
151699 [DBG] Trying serial port FLIP_SIMERETA at \\.\COM3
151699 [DBG] Using  serial port FLIP_SIMERETA at \\.\COM3
151699 [RPC] Starting RPC session...
151703 [RPC] Failed to start RPC session: Failed to begin CLI session: A device attached to the system is not functioning.
151703 [REG] Device initialization failed: Protobuf session error: Failed to begin CLI session: A device attached to the system is not functioning.
151703 [RPC] Failed to start RPC session: Failed to begin CLI session: Failed to begin operation: A device attached to the system is not functioning.