Flipper-XFW/Xtreme-Firmware #269: Change to Xtreme settings introduced bug, RGB Backlight setting affected

oltenxyz, 2023-05-29:
### Describe the bug.

The new setting "Charge Cap" works as intended, but now the RGB Backlight setting is broken.

I don't have the mod so I can't confirm, but the setting might still work by just clicking on the "Charge Cap" setting instead of "RGB Backlight"

Running XFW-0047_29052023.

### Reproduction

1. Go to Xtreme Settings -> Misc -> RGB Backlight
2. Try to press OK to configure -> won't work
3. Move up to "Charge Cap" -> press OK -> "RGB Backlight" dialogue is shown

### Anything else?

Would have loved to just PR the fix, but I wasn't able to identify the fix so far.

ClaraCrazy, 2023-05-29:
i blame whoever merged this without talking to me about it. consider it fixed in the next update.

ClaraCrazy, 2023-05-29:
in fact, its gonna be in a hotfix in a few minutes.