Flipper-XFW/Xtreme-Firmware #273: ESP8266 Deauther not working anymore

iter8tor, 2023-05-29:
### Describe the bug.

On 046 and 047, the ESP8266 Deauther App crashes on startup.

### Reproduction

1. Open App
2. Module's power led lights for less than a second.
3. Power led goes off
4. The Flipper crashes.

ClaraCrazy, 2023-05-29:
outdated app, author abandoned it. we cant maintain 100+ apps

pryh4ck, 2023-06-01:
that sucks lol i literally just bought an 8266 to try it out vs Marauder.. but it may be ok, i got this tiny ass litle kit with an esp-01s and of all things it came with an infared sender and receiver.. so maybe i can figure out how to solder that properly and use it, then make an IR blaster since i can never catch rabbit labs in time before he goes out of stock..

saboaua, 2023-06-02:
I had it working with the d1 mini, I did the update and it also broke. With the previous version had the same issue, I deleted the apps and apps_data folder and got it working. This time without any luck.

closed 2023-06-03 by github-actions[bot]