Flipper-XFW/Xtreme-Firmware #274: The new web updater is unreachable

InToSSH, 2023-05-29:
### Describe the bug.

I noticed starting v47 you are using a new web updater `https://flipper-xtre.me/update`. 
It is unfortunately unreachable with an error from Cloudfront `Error code 523`

### Reproduction

1. Go to realeases
2. Release v47
3. Click Web updater (Chrome)
4. See Cloudflare error

Willy-JL, 2023-05-29:
yes, the web hosting is down due to power outages in some datacenters. just have to wait a bit

InToSSH, 2023-05-29:
That was a super fast response. Thank you, I will wait 😄 I won't close it just now so others can see it in case they have the same question.

ClaraCrazy, 2023-05-30:
back

closed 2023-05-30 by ClaraCrazy