Flipper-XFW/Xtreme-Firmware #277: SubGhz frequencies relock.

Anatejl, 2023-05-30:
### Describe the enhancement you're suggesting.

Hi, original FW has region restriction for transmitting certain waves, all CFW removes this.
How to assign region manually, thus restrict transmission of subghz?
https://docs.flipperzero.one/sub-ghz/frequencies

### Anything else?

Thanks in advance,
Alexandr.

Willy-JL, 2023-05-30:
There is absolutely no point in doing this. If the firmware allows changing the region lock, no one would care (especially law enforcement, if that’s what you’re worried about) if you personally disabled that and kept it restricted.

closed 2023-05-30 by Willy-JL

Anatejl, 2023-05-30:
Sure, but if you can't transmit something to pull law enforcement, they won't care about you. Clearly not to misdirect after illegal action. 
That was the meaning of question. 
Guess then you'd suggest to rollback to OWF, I understand.

ClaraCrazy, 2023-05-30:
You dont get the point, do you? 

Its not illegal to own a device that can transmit there. Just like its not illegal to own a gun, yet murder is. So is actually transmitting. 

And even in OFW you can bypass these restrictions. Theres just no point in having them other than pure marketing and certifications, which they as device manufactures want, but we as code distributors could not care less about