Flipper-XFW/Xtreme-Firmware #278: After installing the new update Flipper Zero just doesn't work anymore

jaochoo, 2023-05-30:
### Describe the bug.

After installing the new update (downloaded .tgz and install via qFlipper) 
Flipper Zero is unable to connect to the computer, screen entirely shut off 

Tried to force restart (left+back button) the light flashes but nothing changed.

Please do help me.

### Reproduction

1. Update the version with .tgz file via qFlipper
2. version updated
3. Flipper immedeatly bricked

jaochoo, 2023-05-30:
after leaving it for couple of minute, I tried force reboot again, this time it went to Update&Recovery Mode: DFU started

Willy-JL, 2023-05-30:
do you happen to have a devboard? would be useful to see some logs

jaochoo, 2023-05-30:
After recovery, the official firmware was installed. I think I will try to install the update again.

Willy-JL, 2023-05-30:
yeah, that was what i was gonna tell you to do after you gave some logs if possible. dont think you can get logs withthose boards, so then just go ahead and reinstall. might want to also use the repair option in qflipper (go into dfu mode and plug in then) before reinstalling xtreme

jaochoo, 2023-05-30:
@Willy-JL Thank you for your assistance, anyway!

jaochoo, 2023-05-30:
The update installation was completed and my Filpper now works fine. I suppose it was an error on my side. Sorry for the inconvenience. I will close the issue now.

closed 2023-05-30 by jaochoo