Flipper-XFW/Xtreme-Firmware #281: Bluetooth “ beacon Spam tool”

BoomSec, 2023-06-01:
### Description of the feature you're suggesting.

Do you know the dstike deauth watch? Essentially that but Bluetooth. Like you can already do it for Wi-Fi networks, but I can imagine someone trying to pair a their fake AirPods just to find a bazillion different Bluetooth devices that they’ve got a train scroll through to find their AirPods rip-offs.

### Anything else?

I just  think it’ll be funny

closed 2023-06-01 by ClaraCrazy

pryh4ck, 2023-06-01:
i thought marauder could already spam bluetooth?

ClaraCrazy, 2023-06-01:
idk tbh. I never tried. Dont see the reason in spamming. but if it can, even more reasons to not work on this as it already exists

pryh4ck, 2023-06-01:
i know you guys arent app makers, but it would be a cool pen test tool if you could somehow add to badKB a bluetooth deauth, that used a bluetooth arduino module or even and esp then spam with a bunch of copies of the device you kicked so when they goto reconnect it, they see a ton, and anyone they choose unless it happens to be theirs, they connect to you instead and then similar to nrf you can choose a badkb script.. so basically nrf but with bluetooth and not 2.4ghz... i dont know if this would even be possible...  a sort of bluetooth jacking like sorts

ClaraCrazy, 2023-06-01:
flipper is a slave, not a master. it cant see devices, it only is seen. therefore we cant deauth, and we wont work on apps exclusive to hardware mods, as only <1% will ever be able to use it, we are only two unpaid people doing this in our free time for some shekels of donations and need to please the many, not the few.