Flipper-XFW/Xtreme-Firmware #283: Pin code changed

Painfir3, 2023-06-03:
### Describe the bug.

I just updated to the new firmware through bt off my phone and when my flipper restarted my pin did not work it seams like it changed plz help

### Reproduction

Update with an old pin and that might do it I'm not sure

### Anything else?

Can't get into my flipper till I get home tonight to while it and start over

ClaraCrazy, 2023-06-03:
pin does not change. tested and works fine. nobody else reports such issue either.

Painfir3, 2023-06-03:
OK I'll see if something happened diffent but for some reason after the
update to the new version of xtreme my pin is not working also I wanted to
say still love the firmware just hope I don't have to wipe it and start
over but if needed I can do that

ClaraCrazy, 2023-06-03:
unless you deleted the lock file, your pin wont have changed. You clearly done *something* wrong, otherwise you wouldnt be here. I use a pin, quite a few users in our Server use a pin, nobody else had issues. 

You can either guess your correct pin, or run the [pin-reset-combo](https://docs.flipperzero.one/basics/control#c_9ya), but keep in mind this would also reset all security applications such as U2F.

closed 2023-06-03 by ClaraCrazy

Painfir3, 2023-06-03:
Thanks and I just did a factory reset but do to there's custom firmware it
only erased my pin and some small gui setup nothing else changed weird but
I got into it easy while I'm not home and thanks for the fast reply I did
not expect anyone to reply. It was just weird that my pin which I made real
easy lol stopped working after an update but thank you alot

Willy-JL, 2023-06-13:
you need to forget the devices, from both flipper and the phone, from both the mobile app and the phone settings.

Painfir3, 2023-06-14:
It was all cleared up when I force factory reset my device and due to it was custom firmware on it all it did was erase the pin and booted up with no pin and all my data was available and fine. So that might be a security flaw that I got in without a pin just had to hold some buttons and factory reset it lol but it's all good. No further comments needed on here thanks everyone.