Flipper-XFW/Xtreme-Firmware #284: Keelog versus Cnt

8um1c4, 2023-06-03:
### Describe the bug.

Iºm trying to reproduce a KeeLoq signal, but the cnt counter doesnºt increment.
In the file I donºt see where to store or change the cnt value. Is it a bug or a feature?

### Reproduction

Yes, sending signal.

ClaraCrazy, 2023-06-03:
Hey there!

This happens if the manufacturer is not known. In that case we obviously cant generate new packets to send, and cant increment any counter. 

Having support for keeloq in general does not mean we can use every keeloq signal that exists, sadly.

closed 2023-06-03 by ClaraCrazy

8um1c4, 2023-06-03:
But Iºve keelog signals recorded that have associated the counter and increase it. I donºt see where is it stored... Can you help?