Flipper-XFW/Xtreme-Firmware #286: Battery indicator style resets itself to default after reboot.

Anatejl, 2023-06-03:
### Describe the bug.

After reboot any different style of charge bar resets to one with number in it.
[Video](https://cdn.discordapp.com/attachments/368580551445315586/1114666507729911928/IMG_0591.mov)

Pics: 
[image]
[image]

### Reproduction

1. Set different battery icon style.
2. Reboot

### Target

X48

Willy-JL, 2023-06-03:
can confirm this, will be fixed soon

Anatejl, 2023-06-03:
Thank you! 
Can I ask a bit about backstage. Saw your commit, is it because of mismatch of variable types? 
As for fix, uint32_t is now in header instead of settings.c. 
It affected in the way, that changes to the battery wasn't saved, and of doing so is being reset on reboot?

Willy-JL, 2023-06-03:
storing the variable as an enum type apparently caused issues when going to save that value, instead of saving the enum value as 0-6, it overflowed somewhere and got to 65k+ or something like that. you can open the settings file and see for yourself. it resets to the default because when reading the setting it reads that 65k value, which is out of bouds of the 0-6, so it rounds down to 6, which happens to be the bar percent and also the default value

closed 2023-07-04 by Willy-JL