Flipper-XFW/Xtreme-Firmware #288: Flipper is not visible in iPhone Bluetooth

T0X1Cx, 2023-06-04:
### Describe the bug.

I can't see my Flipper Zero on iPhone 11 after installing the latest update.

### Reproduction

1. Update last release

Willy-JL, 2023-06-04:
not in bluetooth settings. use the flipper mobile app

closed 2023-06-04 by Willy-JL

ndases, 2023-06-15:
Same case here... even using the mobile app directly
[image]

Willy-JL, 2023-06-15:
no, that is a completely different issue. you came from a previously paired mobile app and installed a recent xtreme. with v47 and v48 the bluetooth code changed, you need to pair again. forget flipper both in your phone's settings and from the mobile app, and also on flipper from settings > bluetooth.

ndases, 2023-06-16:
Thanks mate, worked fine