Flipper-XFW/Xtreme-Firmware #289: windows computer will not show flipper zero as bluetooth device when trying to use Bad KB

EllieEx, 2023-06-05:
### Describe the bug.

Device shows on iphone but will not show on computer even when i try using it as a bluetooth controller in the remote app nothing shows

### Reproduction

1. turn on
2. open Bad KB foler
3. pick ducky file
4. wait for connection to start
5. NOTHING EVER SHOWS TO CONNECT TO

### Target

need to be able to connect to windows laptop via bluetooth

ClaraCrazy, 2023-06-05:
shows fine for 6k+ people in discord and the 200k people that use the fw. sounds like a skill issue to me

closed 2023-06-05 by ClaraCrazy

EllieEx, 2023-06-05:
i figured it out, laptop doesn't support bluetooth (BLE) i'll have to get a bluetooth dongle

ClaraCrazy, 2023-06-05:
yeet.