Flipper-XFW/Xtreme-Firmware #290: Stuck on a screen

EvilWabbit, 2023-06-05:
### Describe the bug.

Hi
The Flipper is frozen on this screen..
[image]
I'm a total noob on the Flipper and this is my first time using Xtreme so excuse me if this is a stupid question but how do I unfreeze this screen?

### Reproduction

I'm sorry but I can't remember how I got here.

ClaraCrazy, 2023-06-05:
Not our app, not our issue. tell the app author

closed 2023-06-05 by ClaraCrazy

EvilWabbit, 2023-06-05:
Again, noob here. I thought I was in Xtreme. What app am I in?

ClaraCrazy, 2023-06-05:
xtreme is a firmware... not an app. idfk what app you opened lmfao

Willy-JL, 2023-06-05:
That is zero tracker, a silly music app. What happened is you disabled volume, so the app can’t play music and is stuck. Just hold back and left to restart

EvilWabbit, 2023-06-05:
That app downloaded with Xtreme. I am a doctor not a coder. I just see a frozen screen. Not a question of using my brain. You still chose to be shitty to someone asking for help. Next time someone comes to me with an illness I must remember to treat them like shit for not having studied medicine.

ClaraCrazy, 2023-06-05:
let me fix your analogy *real* quick.

if they come over with a pill from a different country and complain that it doesnt work, with no packaging or information as to what it is or what its supposed to do, then yes please tell them that they are stupid, bc they might very well be.