Flipper-XFW/Xtreme-Firmware #294: XFW-0048_03062023 Update. Tamagotchi not working anymore.

zcbml88, 2023-06-07:
### Describe the bug.

There is a blank screen when I load and try to play Tamagotchi.. The Flipper freezes and the only way to get it working again is to hard reboot.

### Reproduction

1: Launch Game
2:Flipper freezes

### Anything else?

Does anyone else have this problem?

ClaraCrazy, 2023-06-07:
not our app, open an issue with the creator of that external app plwase

closed 2023-06-07 by ClaraCrazy

Apok76, 2023-06-07:
Couldn’t this be along the same lines as the Xtreme firmware not Syncing correctly a couple of versions ago. You fixed that in the firmware. Or am I mistaken?

-Adam Cloud

ClaraCrazy, 2023-06-07:
? this makes no sense lol.

This app is not written by us, unleashed & rm have the same issue with it, bc the app developer doesnt work on it.

We develop a pentesting firmware, not cringy ahh games for it, so I really dont care about it not working. If it wont be fixed by the dev, the app will just be removed

Apok76, 2023-06-07:
Thanks for the explanation. I was wrong once before. Thanks for number 2.

Love the firmware. You rock.

-Adam Cloud