Flipper-XFW/Xtreme-Firmware #296: Open Sesame implementation

analyserdmz, 2023-06-07:
### Description of the feature you're suggesting.

Hey Flippers!

I was thinking if there's a way to implement [OpenSesame by Samy](https://github.com/samyk/opensesame) into the Xtream Firmware.

The way he shifts bits reduces drastically the duration until a right code is found.

### Anything else?

Here's a [video description](https://www.youtube.com/watch?v=iSSRaIU9_Vc&ab_channel=samykamkar) of how it works. Great thinking, great application of math and overall a great project!

sfjuocekr, 2023-06-24:
There are already resources to be found for this over here: https://github.com/UberGuidoZ/Flipper/tree/main/Sub-GHz/Garages/deBruijn

PuffCuh, 2023-09-23:
> There are already resources to be found for this over here: https://github.com/UberGuidoZ/Flipper/tree/main/Sub-GHz/Garages/deBruijn

Thank you

hotdjdave, 2023-10-02:
None of the deBruijn apps work. I've tried using known codes.

htotoo, 2023-10-04:
The saved files works, but only on prehistoric gates, and thats not the implementation's fault, it is how not so old gates works.

closed 2023-10-31 by Willy-JL