Flipper-XFW/Xtreme-Firmware #297: Frequency settings not applied

lordrak007, 2023-06-08:
### Describe the bug.

When i tried change setting in XFW app, nothing happens, sub GHz still hopping and showing all extended freaquencies.
Restart Flipper Zero does not working

### Reproduction

Change frequencies setting in Protocol section
Go to Sub GHz menu and try hopping or scan specifiec frequencies. There are all frequencies.

Willy-JL, 2023-06-08:
... that menu adds *new* frequencies... they show up at the end of the frequency list in the subghz app, after the default ones. if you only want to see your custom ones, disable "use defaults". note that defaults will still be there if yoiu only add static and not hopper frequencies. need to add at least 1 hopper and 1 static custom frequency in order to hide the defaults.

closed 2023-06-08 by Willy-JL

lordrak007, 2023-06-09:
Ok, now i understand. It is possible to write these information to manual please?

Willy-JL, 2023-06-09:
> And is there way to use only default frequencies?

read my comment again, but slower

lordrak007, 2023-06-09:
:) There is probably misunderstand. Default frequencies i mean locked by default by region. Not all standard enables ones.

Willy-JL, 2023-06-09:
no, region restrictions have been removed because they make no sense. since it was such a simple option to lock or unlock them, we removed that option because it would make no difference in an investigation since it is so easy to change

lordrak007, 2023-06-09:
Ok, thank you