Flipper-XFW/Xtreme-Firmware #301: Had a crash with error Ive never seen before

pryh4ck, 2023-06-10:
Just updated to 48 

Very shortly after it was done I picked it up and it said:

** Flipper crashed but has been rebooted
       ST(R) Copro(R) HardFault**

I haven't seen it sense but im not sure what it is or could mean. Maybe I'll try to flash the update again and reproduce it and if it does it again send you the logs? 

Also when you're looking for logs you want me to turn logging option to debug right?

Willy-JL, 2023-06-10:
thats not something we or you can fix. sorry to break it to you, but your flipper has a faulty chip. that crash is caused by deep sleep not working correctly together with bluetooth on some chips. for now if it happens again and it bothers you, you can disable deep sleep (settings > system > sleep mode = legacy), or just endure it for now. flipper devs and stm are working together to get this fixed via software, hopefully it will get better soon

closed 2023-06-10 by Willy-JL

pryh4ck, 2023-06-10:
oh perfect that's just my luck.. probably won't even replace the flippers effected by this either if they can't fix it through software... idk maybe they will.. for now im just gonna turn off deep sleep its not like Its extremely beneficial... I mean, its nice that it lasts that long (at the expense of your hardware going faulty).. well thanks for the reply 🙏