Flipper-XFW/Xtreme-Firmware #306: BADKB favorites

Retaliation357, 2023-06-19:
### Description of the feature you're suggesting.

In the latest firmware I know there is an option to Favorite/Pin actions like badkb payloads, folders, etc. There should also be an option to Pin certain payloads in folders for Badkb to the top instead of having to search for a specific payload. So if i have a folder with 10 payloads, Id have an option to hold down the OK button for a 3s on a single payload and be able to pin it to top with a little pin icon at the end of the title of the payload. Maybe should also be implemented on RFID, Sub-GHZ, NFC, etc if not too difficult.

Willy-JL, 2023-06-19:
Go to homescreen, press down, then use right and left to navigate to the type of file you want. Find the file or folder you want, press (or hold) ok on it, click pin. Now exit from the menu, pressing down will show your favorites.

closed 2023-06-19 by Willy-JL

Retaliation357, 2023-06-19:
I was referring to that functionality being added into badkb itself. Because I already have a bunch of other things added to my favorites. So my favorites/pins are already a mess itself and I was thinking about it being added into the other apps as well so it can be more organized.