Flipper-XFW/Xtreme-Firmware #309: BadKB: Bluetooth not working

lRoberth, 2023-06-19:
### Describe the bug.

The device does display on nearby BT devices, but on connected, the screen shows "Connect to a device". If I click Run, it stays on "Will run on connect. It's like it is transmitting the BT connection from FZ to devices, but FZ is not responsive when a connection is established. Tried it on a laptop and iPhone.

### Reproduction

1.- Open BadKB
2.- Load file
3.- Go left for config
4.- Set connection BT
5.- Set a random device name
6.- Clicked Randomize BT MAC
7.- Successfully connect device to FZ BT connection
8.- Script cannot be executed as FZ says there's no connection

### Target

iPhone

Willy-JL, 2023-06-19:
Fixed in dev branch

lRoberth, 2023-06-20:
> Fixed in dev branch

Sounds great ,thank you. Also I think I found a sort around. If you go back to script selection, and select it again, it now detects the BT connection and the script runs successfully.

Willy-JL, 2023-06-20:
yes, thats exactly whats fixed in dev branch. along with a huge rework of badkb as a whole

closed 2023-07-04 by Willy-JL