Flipper-XFW/Xtreme-Firmware #314: Ducky script creator

Zarcolio, 2023-06-24:
### Description of the feature you're suggesting.

Last week, I was at an hotel. It had a TV and I had brought my Flipper along. Hooked it up through USB and surely the TV showed a message it had detected an external keyboard 🥳
But than the process halted a bit as I have no easy way to create and test a Ducky script with this TV. Wouldn't it be nice if the was an app to easily create a Ducky script and test it directly? It would be easy to select all kinds of key combos and if you were to write text, it translates it into STRING or if dead keys are detected it uses ALTCODE, etc.
I hope this is doable 😇

ClaraCrazy, 2023-06-24:
no

Zarcolio, 2023-06-27:
For future references, I've created a webpage that can do this on your mobile: https://github.com/Zarcolio/Duckyscript/tree/main/Ducky%20script%20creator

ClaraCrazy, 2023-06-27:
Lets not stay dependant on other devices. Willy has shown interest in creating a flipper app, so I'll re-open this.