Flipper-XFW/Xtreme-Firmware #316: im new an not skilled enough yet

jayuserdrugabuser, 2023-06-29:
### Description of the feature you're suggesting.

i would like to be able to record the buttons pressed on my flipper so it would allow for me to easily  make automated processes or tasks  sequences an save it N have flipper play back what iv done in the last 120 seconds so satarting from the main menu it could record an play back the sequence of inputs used to get to an app set up my configuration if any an start the process  being new im ignorent  to how taxing this is for a flipper an how much resorces it would use up or the demand in ramand memory... so even if u could just help by letting me know if im asking to much from flip flops processing ability that would help thank u

ClaraCrazy, 2023-06-29:
yeah thats absolutely not doable

closed 2023-06-29 by ClaraCrazy

jayuserdrugabuser, 2023-07-07:
Thanks for the reply an for not laughing at my idea im learning more N more each day an with so many different paths to choose form languages to learn an hardware choices an my giant budget of hopes and dreams lol an fastest pair of running shoes lol just kidding ! swim would pull the fire alarm first then hit the electronics depo at swims favorite oversized  mart of some kind with a nylar backpack so a ghost can casualy walk thu security…jkjk…what im tryin to say is I realy do enjoy ur firmware over all the others an the nsfw aspect flipper not some lil richboy toy last request tho can u help make a fapp were I can use the gpio pins to mount a vapepen coiltank battery combo ill build the hardwear it gunna be 0.6ohm coil for salt nic so I just need an app to make it fire 🔥 so like 1wire #17 an grab the sorce code from voopoo.com an togather will turn flipper into PUFFER with a pufferfish logo? Any way thanks an ur super cute <”o”>    ------{project PUFFER}------ Sent from Mail for Windows From: Clara KSent: Thursday, June 29, 2023 10:09 AMTo: ClaraCrazy/Flipper-XtremeCc: jayuserdrugabuser; AuthorSubject: Re: [ClaraCrazy/Flipper-Xtreme] im new an not skilled enough yet (Issue #316) yeah thats absolutely not doable