Flipper-XFW/Xtreme-Firmware #317: BadKB Bluetooth sending inconsistent keystrokes

SiWarlock, 2023-06-29:
### Describe the bug.

When executing a script using the wireless/bluetooth mode for BadKB, the flipper sends inconsistent keystrokes. I am using the Party Parrot for OSX script and when it gets to the line:
STRING curl https://ascii.live/parrot &

75% of the time it will either type:
curl https://ascii.live/parr 
or:
curl https://ascii.live/parrottttttttttt

Maybe 1 out of 4 times does it actually get it right. 
When running the script in USB mode, it executes flawlessly. 

Here is the complete script:

REM Title: Party Parrot for OSX
REM Author: Jeffrey Koopman | JKCTech
REM Description: Shows a pretty party parrot!
REM Target:	macOS 11.0+
REM
DEFAULTDELAY 250
GUI SPACE
STRING terminal
ENTER
DELAY 1000
STRING curl https://ascii.live/parrot &
DELAY 1000
ENTER

I should also note, that the bluetooth connection process is not smooth at all and seems to be buggy. Unsure if this is related or not.

### Reproduction

1. Select BadKB from the menu
2. Select Party_Parrot_OSX from the menu
3. Navigate to config menu
4. Select bluetooth connection
5.  Set random name and mac address
6. Connect to flipper zero from Mac in bluetooth menu
7. Execute script from flipper by pressing 

### Anything else?

Using FW: XFW-0048 03-06-2023

ClaraCrazy, 2023-06-29:
not an issue with the firmware, but with the bluetooth connection between your devices. The code does exactly what its being told, if your device cant keep up with the commands, thats not on us

closed 2023-06-29 by ClaraCrazy

Willy-JL, 2023-06-29:
FWIW however, the delays between keystrokes have been increased in dev branch, so this will probably get better with the next update

SiWarlock, 2023-06-29:
> not an issue with the firmware, but with the bluetooth connection between your devices. The code does exactly what its being told, if your device cant keep up with the commands, thats not on us

If that is the case, then why would there be instances where I have repeating trailing characters? If it were only failing to complete the keystrokes that makes sense, but it seems to be that it sometimes repeats keystrokes that are not part of the script at all.

ClaraCrazy, 2023-06-29:
why do you ask me? Its your device.

If it were an issue with the firmware, tell me why that exact code works fine for over 100k other people, and in two other firmware that asked us to use it

SiWarlock, 2023-06-30:
> FWIW however, the delays between keystrokes have been increased in dev branch, so this will probably get better with the next update

Could you point me to the files that determine the delay between keystrokes when using bluetooth badKB? It seems the speed at which the keystrokes are being sent is too quick for my device to register them accurately. I don't have my other mac to test with at the moment, but I will try and duplicate the issue this weekend.

Willy-JL, 2023-06-30:
It’s not something you can change freely, it needs a recompile. Just wait for the update

SiWarlock, 2023-06-30:
> It’s not something you can change freely, it needs a recompile. Just wait for the update

Appreciate the info, thanks. 
After some messing around, I confirmed that it was indeed the speed of the keystrokes causing the issue. 

For anyone else experiencing this issue, I was able to get around this by adding STRING_DELAY 100 (edit to your desired delay) and this successfully added a delay between each keystroke.