Flipper-XFW/Xtreme-Firmware #318: Morse code SubGhz Receiver/Sender

xAstroBoy, 2023-07-02:
### Description of the feature you're suggesting.

Just a small idea, since the SOS app for subghz with morse code is a cool feature, maybe the app should be able to both receive and decode morse code (with the option to see it raw) and send it as well.

ClaraCrazy, 2023-07-02:
thats literally just the subghz chat but in annoying...

ClaraCrazy, 2023-07-02:
theres a reason why I declined the sos app in the first place, so adding more to a declined idea wont go anywhere

closed 2023-07-02 by ClaraCrazy

xAstroBoy, 2023-07-02:
> theres a reason why I declined the sos app in the first place, so adding more to a declined idea wont go anywhere

oh, i just saw the #182  and thought it was a cool feature.