Flipper-XFW/Xtreme-Firmware #319: Sound Still Plays

Ghdhdhdh, 2023-07-03:
### Describe the bug.

When I turn off sound by clicking up on the home screen and turning the volume to 0,  the volume in games is still apparent, but not in apps list NFC

### Reproduction

1. Turn on the flipper
2. turn the volume to 0
3. Launch Doom

### Target

N/A

### Logs

```Text
Not Needed
```

Willy-JL, 2023-07-03:
complain to the app developers that dont respect how sound works everywhere else on flipper. we dont make those apps

closed 2023-07-03 by Willy-JL

Ghdhdhdh, 2023-07-03:
> complain to the app developers that dont respect how sound works everywhere else on flipper. we dont make those apps

Thanks, Thought it was the firmware.