Flipper-XFW/Xtreme-Firmware #327: Discord issue

Thesek2008, 2023-07-10:
Hi i got banned from your discord only because i have 2008 in my name the real thing is i and my brother use the same pc and he thought its funny to add 2008 to my name can you guys please unbann me thanks

i know normaly thats not the right place for that

My dc thesek2008
i can canget it in like 15 days

Willy-JL, 2023-07-10:
The discord is 18+. You are clearly not 18+, your username says that by itself. Also if I remember correctly you acted like a fool and got yourself banned. The ban is staying for another 3 years until you’re 18 buddy

closed 2023-07-10 by Willy-JL

Thesek2008, 2023-07-10:
why please i like the firmware please i know normaly not but can you unbann me that i can read the discord messages i like the whole project and the firmware