Flipper-XFW/Xtreme-Firmware #330: NFC changes not merged in

craftbyte, 2023-07-18:
### Describe the bug.

It seems like the NFC refactor that happened in OFW didnt get through to here. This make building Seader not work.
https://flipc.org/bettse/seader?branch=main&firmware=xtreme.

It seems like FuriHalNfcDevData does not contain all the fields that OFW does.

### Reproduction

Try building Seader for https://flipc.org/bettse/seader?branch=main&firmware=xtreme

Willy-JL, 2023-07-18:
Pressed close by accident. Anyway those errors are easily fixed, just means that we have to include the app ourselves with the necessary tiny modifications

Willy-JL, 2023-07-26:
removed the WIP nfc felica stuff, so this is now "fixed", apps are compatible as is

closed 2023-07-27 by ClaraCrazy