Flipper-XFW/Xtreme-Firmware #331: Auto-save function in Sub-GHz

ventusz, 2023-07-20:
### Describe the enhancement you're suggesting.

It would be nice to have an auto-save function in the Sub-GHz app.
My imagination is that there could be three settings:
- Off: default, that does not save any signals automatically
- On: which saves all recorded signals
- When memory is full: which only auto-saves when the recorded signals would exceed the 55 limit.

The latter could maybe work FIFO. Saving the oldest then removing it from the list so there is always place for the 55th signal.

Also please consider a save all when exiting feature so that when you don't have time to crawl your way through all the bin_raw signals right there you could just exit the sub-ghz app which then prompts you to choose whether you want to save all or not.

### Anything else?

Yes, I love your firmware, by far it's the best for me among the others. :)

ClaraCrazy, 2023-07-20:
> Off: default, that does not save any signals automatically
> On: which saves all recorded signals

sure

> Also please consider a save all when exiting feature

Not a fan, probably not

ventusz, 2023-07-21:
Thank you! Looking forward to it.

h0us3c4ll, 2023-12-03:
would be nice feature , maybe it also autosaves the timestamp ?

cleanunicorn, 2024-01-07:
Creating a file structure that includes the timestamp is an easy way to sort the files while retaining the chronological order.

Willy-JL, 2024-01-24:
> Creating a file structure that includes the timestamp is an easy way to sort the files while retaining the chronological order.

not an implementation of the issue described here with autosaving, but referencing what you say here with 411a65a42e3dde13dd037c1a6d00130892246f60 you can now choose to have timestamps before in filenames. i thought i remembered someone mentioning something similar, guess it was here in an unrelated issue. still think its worth mentioning

Willy-JL, 2024-01-27:
Now fully implemented. Saves with timestamp into "Autosave" subfolder when enabled, nothing changes if disabled.

closed 2024-02-02 by Willy-JL