Flipper-XFW/Xtreme-Firmware #336: Experimental Apps Feature

NarmakTwo, 2023-07-22:
### Description of the feature you're suggesting.

In the Flipper Mobile **Beta** app, there is a menu which lets you install apps from a library of applications. When you try to install an app, it insists that you update it using OFW. A good feature would be to somehow bypass this.

### Anything else?

I'm not sure if this is still the way the app behaves because I don't regularly update. If this is outdated and is fixed in a future update please tell me.

ClaraCrazy, 2023-07-22:
Welcome to OFW. They check the api on your flipper. If its not OFW, you wont install apps.

We include all apps anyways so theres no need for their app cataloge

Willy-JL, 2023-07-26:
anyway next release shortly will have same api as ofw, so it should work

closed 2023-07-27 by ClaraCrazy