Flipper-XFW/Xtreme-Firmware #346: NFC Fails to load keys

benjaminkovacevic, 2023-08-01:
### Describe the bug.

The Xtreme and Unleashed firmware both have bugs in reading NFC keys.

It would display:
31/32 keys found
15/16 sectors read

The key it fails to load is key B on sector 1.

The thing is, the key B is correct, moreover its the same as key A which was found sucessfully.

I noticed it only produces this bug when the key is manually added to dicontary.

Once flipper is updated to the original firmware all keys are found.

One more thing to mention is that i got the key with NFC nested attack so the problem might be related to cache.

### Reproduction

Switch on
Get a card with unknown key for one of sectors
Using Flipper Nested get the key
Once its added to dicontary, read the tag again
If not working, install default firmware and it should work

d2thunder, 2023-08-14:
Hi,

Have the same Problem today.
It doenst matter in wich way the key is added to the dict_user.
Its always key b. I cleard both lists, controlled codepage and used curragereturn+linefeed and linefeed anlone - all doenst matter.

Another thing i saw, it thale much morentime to test the keys vs sektors.

In unleashed it works with today firmware, think it was a bug from original fw?

Willy-JL, 2023-08-31:
Judging the name i would assume it is something to do with this commit: a7aef0bfc27541639aa7be0cfc06a2db0ecc8c9a

And it being "key invalidation" that had issues, it would make sense that it randomly stopped working after some time, even if the bug was already there... it sounds like a bug that shows itself later on. But I am no NFC expert.

Gonna mark this as release-pending, comment again if it is still not fixed

Willy-JL, 2023-09-01:
should be fixed in v51. try it and let us know

closed 2023-09-01 by Willy-JL

benjaminkovacevic, 2023-10-12:
Had no problems so far. Thank you