Flipper-XFW/Xtreme-Firmware #347: Installing apps via the official flipper iPhone app

mmoti6, 2023-08-01:
### Describe the bug.

I have a flipper zero with xtreme 27.7.23
And when I connected to the official flipper app in iPhone I can't install apps 
The app simply doesn't show me the apps that's available 
If i disconnected the flipper i can see the apps

### Reproduction

1.connect flipper zero with xtreme with iPhone in the official app

Willy-JL, 2023-08-01:
blame OFW. they made it as annoyng as possible. basically the version number is off by 0.1... it wouldwork werent it for the number. it is "fixed" in dev branch, either use the dev builds on discord, or wait for an update

Willy-JL, 2023-09-01:
fixed in v51

closed 2023-09-01 by Willy-JL