Flipper-XFW/Xtreme-Firmware #348: Add SubGhz Test Application

cpressland, 2023-08-02:
### Description of the feature you're suggesting.

The main SubGhz application recently had its "test" feature moved into a dedicated app via https://github.com/flipperdevices/flipperzero-firmware/pull/2851, this is missing in Xtreme. This feature was present in 049, but missing in 050.

Willy-JL, 2023-08-02:
And that feature, is completely useless for end users. It’s placebo. It’s only useful for testing the antenna before shipping the flipper to you. You should not use it.

To quote @xMasterX :
actually there is no guide, but this exact carrier test was made for certification in lab mostly

what it does - waits for Carrier signal that you need to transmit on displayed frequency 
Or transmits carrier signal in TX mode
It has different paths, like isolate - which means antenna is disconnected, only small copper trace on board is connected to chip
You can use it to measure flipper's power output in lab
or its receive capabilities on different distances which is also not correct due to how different digital radio systems work, like lab test with carrier signal is not an garage remote in the open air with all the noise and etc

Willy-JL, 2023-08-02:
It’s still in the source code, if you want it for whatever reason compile it yourself and include the debug pack. But there is no reason to do that.

closed 2023-08-02 by Willy-JL

Speedjet2000, 2023-08-02:
I would like to disagree with this! I have used the test feature to tune several antennas attached to an external CC1101 board. The test functionality allows you to sweep the frequency range and tune based on the RSSI dBm.

xMasterX, 2023-08-02:
> I would like to disagree with this! I have used the test feature to tune several antennas attached to an external CC1101 board. The test functionality allows you to sweep the frequency range and tune based on the RSSI dBm.

do you understand that to actually tune antennas you need Vector Network Analyzer at least simple one like NanoVNA?
who said that tuning using background noise rssi is good idea? again, this is incorrect usage of that carrier test

Willy-JL, 2023-08-02:
Also last I heard the test option did not support external antennas / cc1101 modules... who knows what you’re been tuning then 😂

xMasterX, 2023-08-02:
> Also last I heard the test option did not support external antennas / cc1101 modules... who knows what you’re been tuning then 😂

Previous one that was in time when we used different way to make external radio work, was working with external one, but incorrectly, it had paths that was only present on internal one, since we wasn't actually interested in that test,
correct one will be more "basic" cuz there is no different paths on external modules, only one path

Current one in debug folder that needs to be compiled separately working only with internal as it should

Speedjet2000, 2023-08-02:
>Also last I heard the test option did not support external antennas / cc1101 modules... who knows what you’re been tuning then 😂

I can tell you now that the version of the test app I have is working on external CC1101 modules!