Flipper-XFW/Xtreme-Firmware #350: Sub GHz external module RF test app

Speedjet2000, 2023-08-02:
### Description of the feature you're suggesting.

It would be fantastic to introduce a simple app that allows users to test the efficacy of external antennas and RF modules such as the CC1101. 

The option to be able to select an operating frequency and test RX and TX would be awesome. As someone that is interested in the RF capability of the Flipper and the development of antennas this functionality would be super useful.

closed 2023-08-02 by ClaraCrazy

Speedjet2000, 2023-08-02:
Why was this feature request closed?

Willy-JL, 2023-08-02:
did you read the comment?

ClaraCrazy, 2023-08-02:
Clearly he did not

Speedjet2000, 2023-08-02:
Yes I did. And its n to a duplicate!

Sil333033, 2023-08-02:
n to

ClaraCrazy, 2023-08-02:
n to

Willy-JL, 2023-08-02:
n to

Speedjet2000, 2023-08-02:
Seriously! what's the problem guys? I'm new to this and never posted in Github before. I'm super interested in the Flipper Zero and its capabilities So far all I've been met with is hostility and rude comments! why is that?

ClaraCrazy, 2023-08-02:
There wasnt a single rude comment or hostility. I think your mainframe is corrupt and needs a rewrite.

Speedjet2000, 2023-08-02:
So picking up on my typo's and reposting them isn't rude?!!! You guys are supposed to be the experts here and should be encouraging/guiding people. So can we start again?? Please tell me why my feature request was closed. Thanks

Sil333033, 2023-08-02:
I'm sorry. I thought it was a fun joke. This feature request is deleted because it's useless. Read here for more information [https://github.com/Flipper-XFW/Xtreme-Firmware/issues/348](url) (The other issue where everything is explained) Still thanks for the feature request! If you have any other improvements just let us know!

Speedjet2000, 2023-08-02:
I commented on the other post. I don't find it useless at all, I used it just today to test the power output of a UHF amplifier board that I'm currently working on. My request was not to leave the 'Test' feature in but to develop a new app for the Flipper to make it even better 😎

Sil333033, 2023-08-02:
Amplifier boards are always nice. I would recommend a RAW SubGHz file with junk data. I use that myself and i get a pretty nice view of my output power. 
[image]

fry9, 2023-08-02:
That's a really interesting setup you have there! Are you willing to elaborate a bit about the components that you use?

I've been interested in trying SubGHz amplifier boards, and been looking for some information about what to get. I don't have the Rabbit-Labs module but a few other generic external CC1101 modules and antennas. What kind of wideband power amplifier board is suitable? Is it better with one with a more narrow frequency like 1M-2GHz than 5M-6GHz? What is important? And is that last big board another amplifier, so you have two of them changed?

Sorry for all the questions, maybe it's not suitable here.
But, many thanks for any help.