Flipper-XFW/Xtreme-Firmware #351: Request: Xtreme-App

shivagnihotri, 2023-08-03:
Hi there, much honored developers! 💻

I'm so much in love with your Xtreme firmware, that I'd love to ditch the official Flipper app and switch to an official XFW-App where I can update my lovely Flipper with your firmware. 📲

Is it possible for you to develop it for us? 🙏
(Not sure if this request relates to #344)

Greetings from Germany to all hackers out there! 🧙‍♀️

ClaraCrazy, 2023-08-03:
yes, if you find us people that pay for developers to work on this full-time, with mobile experience and the needed skills.

shivagnihotri, 2023-08-03:
I'd donate for this to happen this, of course. 🌻

ALEEF02, 2023-08-03:
> Is it possible for you to develop it for us? 🙏 (Not sure if this request relates to #344)

Not related to #344. That is specifically about preloaded apps.

bip3r, 2023-08-23:
> I'd donate for this to happen, of course. 🌻

Hey.. there is a mobile app "Flipper Mobile App" with which you can update your FZ with Xtreme firmware. Just make sure to download the older version. 1.3.0 works with mine well.
Newer versions dont recognise Flipper with  firmware other than official (at least Xtreme).

Willy-JL, 2023-08-23:
@bip3r it does recognize flippers with other firmware even on newer versions of the app. If you paired but the app doesn't connect anymore, unpair everywhere (flipper settings, phone settings, phone app). If you mean that it says "Dev (random numbers and letters) as the firmware version, that's a bug in the iOS app, but it still works fine.

bip3r, 2023-08-23:
I will check the newest then. After 1.3.0 I did all from scratch and it didnt recognise. Maybe it was already improved.

shivagnihotri, 2023-08-23:
> Hey.. there is a mobile app "Flipper Mobile App" with which you can update your FZ with Xtreme firmware. Just make sure to download the older version. 1.3.0 works with mine well.
> Newer versions dont [...]

Thank you so much for telling me! 🥰

If the official Flipper-App works with updating Xtreme, I guess there is no need to develop a new app and this Issue may be closed. It would be wonderful though, if this is mentioned in the documentation or README somewhere. 🍀

closed 2023-08-24 by shivagnihotri