Flipper-XFW/Xtreme-Firmware #353: Evil portal crashes when file is selected

N-coder82, 2023-08-10:
### Describe the bug.

When I open the Evil Portal app with Wi-Fi Devboard connected (the one from flipperzero.one with Marauder firmware on it), and then select the demo portal, the entire thing crashes, and flipper reboots with "NULL Pointer Deference". I attached logs here because I couldn't upload files in the log section. In the zip there are multiple files.

[evil_portal_logs.zip](https://github.com/Flipper-XFW/Xtreme-Firmware/files/12315802/evil_portal_logs.zip)

### Reproduction

1. Open Apps
2. Navigate to "Wifi" Section
3. Click on"[ESP32] Evil Portal"
4. Click "Start portal"
5. Select "xtreme" in the file selector
6. Click on "xtreme in the file selector"
7. Flipper crashes and reboots with "NULL Pointer Deference"

### Anything else?

I love this firmware! I went from Unleashed for a day or two, then to Rogue Master for a half-hour, and I've been here since, it's by far the most stable and customizable firmware available, and all the features are high-quality and amazing, thank you in advance.

Willy-JL, 2023-08-10:
You even explained the issue yourself. You have marauder firmware on the board. Only marauder will work. If you want to use evil portal, you need to flash evil portal firmware to the board. The esp flasher app will help you, just apps > gpio > esp flasher > flash wifi dev board > flash evil portal

closed 2023-08-10 by Willy-JL

N-coder82, 2023-08-10:
thank you, I didn't realize that you needed a different firmware!

skitzo676, 2023-10-10:
i tried same thing but still doesnt work